epics-base/epics-base #607: Multiple redundant installation of doc files

dirk-zimoch, 2025-02-21:
When building `modules/database/src/std/`, I see this for each of my (cross) architectures:
```
Installing doc ../../../../../doc/image/compress-1.png
Installing doc ../../../../../doc/image/compress-2.png
```
The Makefile contains:
```
DOCS += image/compress-1.png
DOCS += image/compress-2.png
HTMLS += image/compress-1.png
HTMLS += image/compress-2.png
```
Comparing with the HTML installation, I see this only once:
```
Installing html ../../../../../html/./image/compress-1.png
Installing html ../../../../../html/./image/compress-2.png
```
And also when running `make` twice, the `HTMLS` installation only happens the first time while the `DOCS` installation is executed each time.

The difference lies in `RULES_BUILD`:
```
$(INSTALL_DOC)/%: %
	$(ECHO) "Installing doc $@"
	@$(INSTALL) -d -m $(INSTALL_PERMISSIONS) $< $(INSTALL_DOC)
```
vs
```
$(INSTALL_HTML)/$(HTMLS_DIR)/%: %
	$(ECHO) "Installing html $@"
	@$(INSTALL) -d -m $(INSTALL_PERMISSIONS) $< $(@D)
```

When replacing `$(INSTALL_DOC)` with `(@D)`, the installation happens only once, but now preserves the subdirectory structure, installing `doc/image/compress-*.png` instead of `doc/compress-*.png`.

I am not sure how this change of installation directory causes `make` to believe that the files need to be re-installed.

simon-ess, 2025-02-21:
This looks to me like the problem is that we are trying to install `../../../../../doc/image/compress-1.png`, but we are not actually installing it to the correct location. If you look at the automatic variables for that build rule:
```
$@ = ../../../../../doc/image/compress-1.png
$< = ../rec/image/compress-1.png
$* = image/compress-1.png
$(INSTALL_DOC) = ../../../../../doc
$(@D) = ../../../../../doc/image
```
So we are installing `image/compress-1.png` to `doc`, not `doc/image`. So on the next build the file that it is looking for (`doc/image/compress-1.png`) does not exist, and hence it re-makes it.

That said, I haven't looked at the documentation that is installed to see if this is the correct location or not.

simon-ess, 2025-02-21:
I believe it is the incorrect install path:
```
<div>
    <img src="image/compress-1.png">
</div>
```
is in the output.

dirk-zimoch, 2025-02-21:
That would mean that replacing `$(INSTALL_DOC)` with `$(@D)` is probably the correct thing to do.
Will it affect any other files?

simon-ess, 2025-02-21:
It seems that there are four files that use this rule:
```
../../../../../doc/msi.md
../../../../../doc/image/compress-1.png
../../../../../doc/image/compress-2.png
../../../../../doc/ComponentReference.md
```
So the other two would be fine. I didn't see anything that used the second form of the rule, but I wasn't  being super careful.

anjohnson, 2025-06-11:
Thanks to @simon-ess for the investigation!

commit 81ba6a8a (2025-06-11): Fix INSTALL_DOC rule

Fixes #607
closed 2025-06-11 by anjohnson