epics-base/epics-base #616: mingw warnings about order of headers

dirk-zimoch, 2025-03-12:
When cross compiling pvAccess for win32-x86-mingw or windows-x64-mingw on RHEL9, I noticed the following warnings:
```
/usr/i686-w64-mingw32/sys-root/mingw/include/winsock2.h:15:2: warning: #warning Please include winsock2.h before windows.h [-Wcpp]
```
This happens when code calls
```
#include  <epicsAtomic.h>
```
before
```
#include <osiSock.h>
```
The mingw version of winsock2.h emits the warning here:

```
#ifndef _WINSOCKAPI_
#define _WINSOCKAPI_
#else
#warning Please include winsock2.h before windows.h
#endif
```
See:
* [1] https://sourceforge.net/p/mingw-w64/wiki2/winsock2.h%20include%20order/
* [2] https://cplusplus.com/forum/windows/112014/
* [3] https://stackoverflow.com/questions/9153911/is-there-a-difference-between-winsock-h-and-winsock2-h

[3] says: "`windows.h` includes `winsock.h` by default (if `WIN32_LEAN_AND_MEAN` is not defined)."

For MSVC, `VC_EXTRALEAN` seems to work too, which is set in WIN32/epicsAtomicOSD.h, but mingw does not use `VC_EXTRALEAN`. It needs `WIN32_LEAN_AND_MEAN`.

WIN32/epicsAtomicOSD.h spends some effort to `#undef VC_EXTRALEAN` if it had not been set before. Is that necessary (-> @FreddieAkeroyd )?
Maybe we need to do the same for `WIN32_LEAN_AND_MEAN`. Or instead?

FreddieAkeroyd, 2025-03-12:
Hi @dirk-zimoch I think this is resolved by https://github.com/epics-base/epics-base/pull/518 which makes some changes to avoid importing unnecessary definitions, and setting  `WIN32_LEAN_AND_MEAN` is one of the things it does

dirk-zimoch, 2025-03-13:
Hi @FreddieAkeroyd, yes, I think that PR would fix this problem.

closed 2025-06-06 by dirk-zimoch