epics-base/epics-base #620: How to compile only for a target architecture

JaviRP9, 2025-03-24:
Hello, according to the [documentation](https://docs.epics-controls.org/en/latest/getting-started/cross-compile-to-old-x86-linux.html#configure-cross-compilation-in-epics), EPICS-base compiles for both the host architecture and the specified target architecture. I wanted to know if it is possible to set some configuration in the EPICS base to cross-compile only for a target architecture. In the EPICS Teck Talk, I found this [post](https://epics.anl.gov/tech-talk/2017/msg01113.php), but it does not reach any conclusion on how to avoid the host architecture compilation of the EPICS base.

Any idea on how to do it?

ralphlange, 2025-03-25:
That question is something that is better asked on Tech-Talk than here.

The Tech-Talk thread you refer to does have an answer, by the way:
https://epics.anl.gov/tech-talk/2017/msg01116.php
Is that not working for you?

JaviRP9, 2025-03-25:
Hello, excuse me, I will try opening a thread on Tech-talk. In my case, that Tech-talk thread indicates how to compile an EPICS IOC for only one target architecture, but my problem arises with the compilation of EPICS base. Even though I correctly configured the EPICS base for cross-compilation and even used the rules that indicate that it only compiles the specified architecture:
`make build.linux-arm`
The result is that it still also compiles the EPICS base for the host architecture (in my case, linux-x86_64):

[image]

simon-ess, 2025-03-25:
Hello –

From Andrew’s response here: https://epics.anl.gov/tech-talk/2017/msg01119.php it seems like this isn’t really possible.

Is there any reason you cannot just build EPICS base for both host and target architectures and ignore the host architecture?

Cheers,

Simon

Reply to: epics-base/epics-base ***@***.***>
Date: Tuesday, 25 March 2025 at 11:18
To: epics-base/epics-base ***@***.***>
Cc: Subscribed ***@***.***>

image.png (view on web)<https://github.com/user-attachments/assets/e840241c-9f83-4902-a67c-982ff27b8f10>

image.png (view on web)<https://github.com/user-attachments/assets/e840241c-9f83-4902-a67c-982ff27b8f10>

JaviRP9, 2025-03-25:
Thank you very much, @ralphlange and @simon-ess. My intention was to find out if this topic had been addressed in the latest versions of EPICS. In my case, it was solely to optimize the compilation time on my system by avoiding compilation for the host, as I didn’t need it. Thank you very much for your response.

ralphlange, 2025-03-25:
Well, as Andrew pointed out in the Tech-talk thread:
- The EPICS Base build creates tools (from source) that are needed on the host, even for cross-builds.
- In the EPICS Build system, artifacts that are architecture-independent (databases, database definitions, documentation, ...) are generated on the host to avoid generating them multiple times (for each target architecture).

In principle, the Build system could be changed to separate needed and unneeded targets for cross-only setups. I fear, though, that this would require considerable effort and add even more complexity.

Most often, enabling parallel make is improving the build time a lot more. Set `-j16` on a recent machine and don't burn yourself trying to empty your coffee before it finishes.

closed 2025-03-25 by ralphlange

anjohnson, 2025-03-25:
We have separated out most of the tools required by build-hosts from the IOC and other target programs in recent years. The `VALID_BUILDS` Makefile variable can be set in an arch-specific CONFIG_SITE file to tell the build system what kinds of things to build for this target. I thought at one time it was at one time possible to do host-only builds by setting that variable to `Host`, but we don't have a CI build which checks that, and the current 7.0 branch needs several Makefiles to be fixed. Note though that you still need to have a C/C++ compiler for the host, there's no way around that requirement.

I did just work out what the changes are to the 7.0 branch and built darwin-aarch64 as a build-host only (but without the PVA submodules), but cross-building for ios-arm isn't the same as for linux so I might have missed some things. I can turn that into a PR if there is enough demand for this.

mdavidsaver, 2025-03-31:
> it was solely to optimize the compilation time

fyi. many years ago I looked at doing split host/target builds in the context of the epicsdeb work.  Ultimately I decided the modest saving in compile time was not worth the effort to test and maintain yet another build scenario.

If you are interested, the last vestige of this experiment can be found here:

https://github.com/epicsdeb/epics-base/blob/3964bbfe28bef45167ef83d63c767b827c3a8a42/debian/patches/0008-SKIP_HOST-build-option.patch

Note that this patch was last exercised in 2012.  While it applies to Base 7.0, it may not work.