epics-base/epics-base #632: "true" is a C23 keyword

mdavidsaver, 2025-04-15:
`true` becomes a keyword with [C23](https://en.cppreference.com/w/c/keyword).

https://github.com/epics-base/epics-base/blob/7372b4e5bb0a5ab75e5403bbbf131f11ad5639d6/modules/libcom/src/log/iocLogServer.c#L483

Fails with GCC 15.0.1-0.11.fc42 (ships with fedora 42 as the version suggests) which [defaults](https://gcc.gnu.org/onlinedocs/gcc/Standards.html) to `-std=gnu23`.

```
gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE        -fdiagnostics-color  -D_X86_64_ -DUNIX  -Dlinux   -DBUILDING_libCom_API   -O3 -g   -Wall -Werror-implicit-function-declaration      -mtune=generic     -m64  -fPIC -I. -I../O.Common -I. -I../osi/compiler/gcc -I../osi/compiler/default -I. -I../osi/os/Linux -I../osi/os/posix -I../osi/os/default -I.. -I../as -I../bucketLib -I../calc -I../cvtFast -I../cppStd -I../cxxTemplates -I../dbmf -I../ellLib -I../env -I../error -I../fdmgr -I../flex -I../freeList -I../gpHash -I../iocsh -I../log -I../macLib -I../misc -I../osi -I../pool -I../ring -I../taskwd -I../timer -I../yacc -I../yacc -I../yajl -I../../../../include/compiler/gcc -I../../../../include/os/Linux -I../../../../include         -c ../yajl/yajl_parser.c
  make[4]: Leaving directory '/home/runner/.cache/base-7.0/modules/libcom/src/O.linux-x86_64'
  ../log/iocLogServer.c: In function ‘acceptNewClient’:
  ../log/iocLogServer.c:483:14: error: expected identifier or ‘(’ before ‘true’
    483 |         long true = 1;
        |              ^~~~
  ../log/iocLogServer.c:489:25: error: lvalue required as unary ‘&’ operand
    489 |                 (char *)&true,
        |                         ^
```

mdavidsaver, 2025-04-15:
Quick fix with f827bd7e34450ffbfa6e1d724e7164c24117bc2d

mdavidsaver, 2025-04-17:
I'm changing this into a general C23 build failure issue.  (I was overly optimistic that there was only one...)

See #638 for another.

anjohnson, 2025-04-17:
All resolved in https://github.com/anjohnson/epics-base/commit/e13d65f6f56dfff61d5e8c28831c1810977b6a06

I'm just waiting for the last jobs to complete before pushing that to the main 7.0 branch.

anjohnson, 2025-04-17:
Fix e13d65f6f56dfff61d5e8c28831c1810977b6a06 pushed.

closed 2025-04-17 by anjohnson