epics-base/epics-base #633: Any better way to control the execution order of registrars?

konosubakonoakua, 2025-04-16:
I have export multiple registrars in different files.
I need to execute register mapping registrar first, then the other registrars.
Currently I'm using naming convention prefixed by `a_` to `z_` to hack the `blm_registerRecordDeviceDriver` function:
```c
int blm_registerRecordDeviceDriver(DBBASE *pbase)
{
    static int executed = 0;
    const char *bldTop = "/root/epics/iocs/blm-zu9";
    const char *envTop = getenv("TOP");

    if (envTop && strcmp(envTop, bldTop)) {
        printf("Warning: IOC is booting with TOP = \"%s\"\n"
               "          but was built with TOP = \"%s\"\n",
               envTop, bldTop);
    }

    if (!pbase) {
        printf("pdbbase is NULL; you must load a DBD file first.\n");
        return -1;
    }

    if (executed) {
        printf("Warning: Registration already done.\n");
    }
    executed = 1;

    registerRecordTypes(pbase, NELEMENTS(rtl), recordTypeNames, rtl);
    registerDevices(pbase, NELEMENTS(devsl), deviceSupportNames, devsl);
    registerJLinks(pbase, NELEMENTS(jlifsl), jlifsl);
    // here
    runRegistrarOnce(pvar_func_a_devZynqPLReg_register_commands);

    runRegistrarOnce(pvar_func_arrInitialize);
    runRegistrarOnce(pvar_func_asSub);
    runRegistrarOnce(pvar_func_dbndInitialize);
    runRegistrarOnce(pvar_func_decInitialize);
    runRegistrarOnce(pvar_func_devZynqMPMonitorRegister);
    runRegistrarOnce(pvar_func_devZynqPSUART_register_commands);
    runRegistrarOnce(pvar_func_helloRegister);
    runRegistrarOnce(pvar_func_rsrvRegistrar);
    runRegistrarOnce(pvar_func_syncInitialize);
    runRegistrarOnce(pvar_func_tsInitialize);
    runRegistrarOnce(pvar_func_utagInitialize);
    
    // here
    runRegistrarOnce(pvar_func_z_blmCommonRegister);
    iocshRegisterVariable(vardefs);
    return 0;
}
```

ralphlange, 2025-04-16:
Such questions are better handled on the Tech-Talk mail exploder or the EPICS Chat system.

ralphlange, 2025-04-16:
Nevertheless:

I think you're not using the mechanism as intended. Please have a look at the `initHooks` mechanism that allows user callbacks to be run at specific moments during the IOC initialization.
https://docs.epics-controls.org/en/latest/appdevguide/IOCInit.html#inithooks

It's a two-step process. The registrars are expected to _register_ things like the aforementioned init hooks (hence their name). That kind of registration should not require any specific order.
Later, at IOC initialization, initHooks is running the registered callbacks at the specified moments.

closed 2025-04-17 by konosubakonoakua