epics-base/epics-base #664: New number output format(s) for caget & camonitor

anjohnson, 2025-06-03:
I found myself recently wanting to see how much memory an IOC was using. It had the iocStats PVs loaded, but the default output from fetching those PVs with `caget` didn't make the results easy to understand:
```
$ caget `grep :MEM_ /C2/iocinfo/pvdata/sioc2beamqual`
sioc2beamqual:MEM_UPD_TIME     10
sioc2beamqual:MEM_USED         1.2034e+07
sioc2beamqual:MEM_FREE         1.34346e+10
sioc2beamqual:MEM_MAX          1.66578e+10
```

These PVs are doubles, so `caget -f0` is needed to avoid the exponential formatting:
```
$ caget -f0 `grep :MEM_ /C2/iocinfo/pvdata/sioc2beamqual`
sioc2beamqual:MEM_UPD_TIME     10
sioc2beamqual:MEM_USED         12034048
sioc2beamqual:MEM_FREE         13435002880
sioc2beamqual:MEM_MAX          16657793024
```
Like most humans though, I can't immediately see exactly how big those values really are either.

The output options from `caget` include these:
```
Floating point type format:
…
  -lx:     Round to long integer and print as hex number
  -lo:     Round to long integer and print as octal number
  -lb:     Round to long integer and print as binary number
Integer number format:
  Default: Print as decimal number
  -0x: Print as hex number
  -0o: Print as octal number
  -0b: Print as binary number
```

The `ls`, `df` and similar Unix-derived programs often have a `-h` option which rounds and prints sizes in a human-readable format, using suffixes `K`, `M`, `G` and presumably `T`, `P` and `E` when the file-system is big enough.

This issue is a request to add a similar `-lh` option to `caget` and `camonitor` that would round double values to a long integer and print them as a "human" number, giving something like this output:
```
$ caget -lh `grep :MEM_ /C2/iocinfo/pvdata/sioc2beamqual`
sioc2beamqual:MEM_UPD_TIME     10
sioc2beamqual:MEM_USED         12.0M
sioc2beamqual:MEM_FREE         13.4G
sioc2beamqual:MEM_MAX          16.7G
```

The similar option `-0h` could also be added for printing `DBR_LONG` values as a "human" number, although it might not get used as much.

anjohnson, 2025-06-04:
@mdavidsaver Ack, an "engineering" notation format would work too (or in addition); I don't know what the command-line option would be for that though.

> ambiguity

True, we should document which multipliers we use. When you need that kind of precision and don't remember what we give you just don't use `-lh` perhaps?

ralphlange, 2025-06-04:
What I don't quite like is adding complexity to all-purpose tools just for the very non-standard case of
doubles that hold integer numbers of properties that are often expressed as powers of two.

What if I would ask for caget on strings changing font colours if the string can be interpreted as a colour name?

ralphlange, 2025-06-07:
> > What if I would ask for caget on strings changing font colours if the string can be interpreted as a colour name?
> 
> Are you thinking of ANSI terminal escapes in values?

Sorry - I was trying to make a joke. (And failed.)

To me, a formatting option for "doubles that are really integers that are often expressed as powers of 2" is as practical as a formatting option that recognizes color names in string values and prints those words in that color.

mdavidsaver, 2025-06-10:
> Sorry - I was trying to make a joke. (And failed.)

Yup, that one got right past me

> ... doubles that are really integers ...

I agree.  I don't see a strong motivation for complexifying the argument list of `caget` and friends.

What I was suggesting was to change how `caget` prints all floating point values.  However, I am uncertain whether the added code complexity of not being able to use a simple `printf` spec. would be worth the effect.