epics-base/epics-base #666: Template substitution syntax without macros is not respected by dbLoadTemplate nor msi

henriquesimoes, 2025-06-06:
While attempting to load a template in a substitution file without any macro substitution, I found an inconsistency in the `msi` and `dbLoadTemplate` implementation regarding the [documented syntax](https://docs.epics-controls.org/en/latest/appdevguide/databaseDefinition.html#template-file-syntax). More specifically, in the `template-subs` rule, `subs` is documented as optional:

```
substitution-file ::= ( global-defs | template-subs )+
template-subs ::= template-filename '{' subs? '}'
template-filename ::= 'file' file-name
```

However, when attempting to load `foo.substitutions` with the following content

```
file "bar.template" {}
```

we get a parse error:

```sh
$ msi -S foo.substitutions
msi: Parse error, unexpected '}'
  in substitution file 'foo.substitutions' at line 2:
  file "bar.template" {}
```

Similarly, `dbLoadTemplate` doesn't load any records (and doesn't report any error as well):

```sh
$ softIoc
epics> dbLoadTemplate("foo.substitutions")
epics> dbl
epics> 
```

Turning it into a script with `on error break` doesn't cause it to break at all. If other `file` entries exist on `foo.substitutions`, they are correctly parsed, which shows me that it simply ignores the `file "bar.template" {}` silently.

We can workaround this by adding an extra pair of braces:

```
file "bar.template" {{}}
```

That's an awkward syntax, though. I'm opening an issue as I'm not sure if changing `msi` and `dbLoadTemplate` is reasonable, or if it's better to simply drop the `?` from the grammar. As for the second option, I think it would be nice if we could change `dbLoadTemplate` to at least warn it has ignored that `file` directive.

ralphlange, 2025-06-06:
I don't think that macros are related to this issue.

For me,
```
file "bar.template" {
}
```
is declaring a template with no instance. Which should produce no records.

On the other hand,
```
file "bar.template" {
  {}
}
```
is declaring the template and a single instance with no macro expansion, which would lead to the records from that template being created once.

mdavidsaver, 2025-06-06:
@henriquesimoes Which version of Base are you testing with?

> Turning it into a script with `on error break` doesn't cause it to break at all.

I think this aspect was fixed by 1cd141c540c5532f1618c538c3a81a62fc73c639 in Base 7.0.9.

mdavidsaver, 2025-06-09:
> > Turning it into a script with `on error break` doesn't cause it to break at all.
> 
> I think this aspect was fixed by [1cd141c](https://github.com/epics-base/epics-base/commit/1cd141c540c5532f1618c538c3a81a62fc73c639) in Base 7.0.9.

> I'm afraid it actually wasn't. If I change `foo.substitutions` to:

> iocsh Error: Break
> ERROR: Error in ./st.cmd

Perhaps I misunderstood your original statement.  The aspect I referred to is propagation of `dbLoadTemplate()` errors back to iocsh.  Seeing `iocsh Error: Break` tells me that `on error break` is having the desired effect.

henriquesimoes, 2025-06-09:
> Perhaps I misunderstood your original statement. The aspect I referred to is propagation of `dbLoadTemplate()` errors back to iocsh.

Oh, okay. From that perspective, it indeed took effect. The thing is that, when there is no instance, there is no error to be propagated to begin with. So the end result kept the same as in my initial report. From your initial comment, I was expecting to see something different in that scenario. ;)

(Using `on error break` was just a way to show that the return value was non-error as well. It turns out your patch is indeed required for that to make sense at all, though).

I hope my answer is a bit clearer now.

mdavidsaver, 2025-06-10:
> I hope my answer is a bit clearer now.

Indeed it is.  Lack of error propagation is a straightforward bug.  What constitutes an error... can be more complicated.

As for the later, I think that

```
file "bar.template" {
}
```

should not be a syntax error, or even a warning.  I think the intent is clearly to load no instance.

A slight nuance, should the existence of `bar.template` be enforced even when no instance is loaded?