epics-base/epics-base #669: Question on libCap5.so

jeonghanlee, 2025-06-23:
@anjohnson

I have a question regarding `libCap5.so`.

Within the build system, this library is compiled once and then installed into two different locations:

* `base/lib/perl/5.36.0/x86_64-linux-gnu-thread-multi`
* `base/lib/linux-x86_64`

While working on a fully "relocatable" EPICS installation, @lerwys and I observed that the `libCap5.so` located in `x86_64-linux-gnu-thread-multi` fails to find its dependent libraries. Running `ldd` confirms this:

```bash
x86_64-linux-gnu-thread-multi (relocatable)$ ldd libCap5.so
    linux-vdso.so.1 (0x00007ffd8c96d000)
    libca.so.4.14.2 => not found
    libCom.so.3.22.0 => not found

```

Conversely, the libCap5.so in base/lib/linux-x86_64 correctly locates its dependencies:.

```
linux-x86_64 (relocatable)$ ldd libCap5.so
    linux-vdso.so.1 (0x00007fff94bcb000)
    libca.so.4.14.2 => /home/jeonglee/gitsrc/EPICS-env-distribution/1.1.1/debian-12/7.0.7/base/lib/linux-x86_64/././libca.so.4.14.2 (0x00007f033bd13000)
    libCom.so.3.22.0 => /home/jeonglee/gitsrc/EPICS-env-distribution/1.1.1/debian-12/7.0.7/base/lib/linux-x86_64/././libCom.so.3.22.0 (0x00007f033bc9d000)
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f033ba98000)
```
Both instances of libCap5.so currently have the same RUNPATH:

```
chrpath libCap5.so
libCap5.so: RUNPATH=$ORIGIN/.
```
My questions are:

- What is the specific purpose of the `libCap5.so` located within `base/lib/perl/5.36.0/x86_64-linux-gnu-thread-multi/`? Is it genuinely required for our setup?

- If it is required, what would be a feasible solution to ensure both libCap5.so files correctly find their dependent libraries (`libca.so` and `libCom.so`) within a relocatable installation?

Thanks,
@jeonghanlee

anjohnson, 2025-06-24:
Hi @jeonghanlee, 

`libCap5.so` is loaded by the Perl `CA.pm` module which provides bindings for Perl programs that want to talk Channel Access. You can disable building these bindings by commenting out the following line in `$BASE/modules/ca/src/Makefile`, which stops them from being provided but would have no other effect on your Base installation:
```Makefile
DIRS += perl
```
If you don't care about supporting Perl in your fully-relocatable installation, stop reading here and do the above. If you want to provide the CA bindings though, read on…

I don't know how much other sites use them, but at APS the Perl CA bindings are essential for many of our production applications. Base comes with a makeBaseApp template which has code for scripts called `cainfo.pl`, `caget.pl`, `caput.pl` and `camonitor.pl` that demonstrate how to use the API, and it also installs the `capr.pl` script, an implementation of the IOC's `dbpr` command that can be run remotely. I find that script really useful at times.

A site may have multiple versions of Perl installed, which would each need their own build of the `libCap5.so` loadable library, so the interface code in CA.pl looks for that library in the `$BASE/lib/perl/<perl-version>/<perl-architecture>` directory. That lets users run shared Perl CA applications on whatever Perl version is provided by the OS their machine is running; at APS we currently have 4 different versions installed in our NFS-shared production area:
```
$ ls /C2/ioc-modules/build/BASE/7.0.6.1-C2-3+HE-007/lib/perl/
5.26.2/  5.26.3/  5.32.1/  5.34.0/  CA.pm  DBD/  DBD.pm  EPICS/  EpicsHostArch.pl
```

It isn't _necessary_ for `libCap5.so` to also be installed into `$BASE/lib/<host-arch>` but the EPICS build system does that automatically, and it would take more work to stop it while still building the loadable library.

If you still want to install the Perl bindings but intend your Base installations to only ever have one version of Perl and one host architecture installed it should be possible to modify the `CA.pm` bootstrap code to load the `$BASE/lib/<host-arch>` copy instead, and to not install the other one at all. The main complication is that the Perl code doesn't know what the EPICS `<host-arch>` string is. As long as the `EPICS_HOST_ARCH` environment variable is set or you always use the `EpicsHostArch.pl` script to determine it and never add private host architectures (some sites add architectures like `rhel8-x86_64` `rhel-9-x86_64` etc.) that would be a relatively straightforward change to make.

I suggest loading the `$BASE/lib/<host-arch>` copy because that doesn't need a change to your `RUNPATH`. Alternatively it would be possible to copy the `<host-arch>` string into the installed copy of `CA.pm`, but again that fixes the installation to run Perl applications on the Perl installation that was used to at build-time.

Modifying a fully-relocatable build to also support multiple Perl versions and architectures would be more work solving the `RUNPATH` problem though. I'd be happy to advise you on any of the above solutions if you do want to support the Perl CA bindings.

HTH,

- Andrew

jeonghanlee, 2025-06-24:
Andrew,

Thank you for the deep and kind explanation.
We don't use Perl, so it is safe to keep it as it is. 
I've learned more than what I thought. Thank you.

Han

P.S. I **never** use the private host architectures, so it is straightforward to me. Here is my own training for the ALS-U team about that host architecture argument.

https://jeonghanlee.github.io/epics-trainings/CH1/01.03.epicshostarch.html

jeonghanlee, 2025-06-24:
I closed it, since I have the brilliant answer.

closed 2025-06-24 by jeonghanlee