epics-base/epics-base #670: Controlled behaviour of the IOC shutdown (follow-up PR #558)

jerzyjamroz, 2025-06-25:
The idea of this ticket is to find out the best solution for handling hardware registry/config modifications upon IOC shutdown in a standardized way, whereas the IOC process is managed by e.g. systemd.

It is a follow-up to the PR #558 design conclusions.

jerzyjamroz, 2025-06-25:
@anjohnson @mdavidsaver , [this line](https://github.com/epics-modules/mrfioc2/blob/e71217d79588ea4dad7d83368af79f0dfae59c45/evrMrmApp/src/drvemSetup.cpp#L981) is never executed upon `SIGTERM`.

anjohnson, 2025-06-25:
It's not just the iocsh `exit` command; in the osi/posix/osdThread.c::`once()` routine we do this, which catches Ctrl-C and other exit paths:

[`status = atexit(epicsExitCallAtExits);`](https://github.com/epics-base/epics-base/blob/f911f8ca3eb94e0a7be3e3ed26166cc63ad8dac4/modules/libcom/src/osi/os/posix/osdThread.c#L414C21-L414C41)

I'm not surprised the `SIGTERM` handler doesn't trigger `atexit()` routines though.

@mdavidsaver says we can't call `epicsExitLater()` directly from a signal handler, so how could we arrange for a `SIGTERM` handler to do something legal that would result in another thread calling that?

jerzyjamroz, 2025-06-26:
Thank you for the explanations. Just for the log, it looks like that for `SIGTERM`.

```
epics>
epics> var atExitDebug 1
Terminated
```

jerzyjamroz, 2025-06-27:
For the log, this subject is extended with #671

jerzyjamroz, 2025-07-01:
Ok, so once the @dirk-zimoch PR https://github.com/epics-base/epics-base/pull/671 is done. Let's evaluate the initial idea so the iocsh command: "initHook`AtShutdown <command>`". The difference here is that one can define the exit procedures in `.iocsh` files, so one does not need to implement it in C/C++ code of the driver and gets an easy and clean way to control it.

jerzyjamroz, 2025-07-07:
For the log: https://github.com/epics-base/epics-base/pull/671#issuecomment-3044353256

jerzyjamroz, 2026-09-04:
I tested `mdavidsaver:sigint-readline` and `dirk-zimoch:exitSig`.
`sigint-readline` (PASSED tests) is a minimal and relatively non-invasive implementation. It supports only `SIGINT`.

`exitSig` (NOT PASSED tests) requires more work, but it aims to provide a unified IOC shutdown mechanism for both `SIGINT` and `SIGTERM`.

In principle, `sigint-readline` could be added as a improvement while `exitSig` is evaluated.

A possible long-term solution would be to combine the two approaches somehow, but `sigint-readline` already gives the required feature.