epics-base/epics-base #684: Compile Error

gtt875, 2025-07-21:
[image]

mdavidsaver, 2025-07-21:
Details please.  Which version of epics-base?  Which environment?  (looks like mingw on windows?)

gtt875, 2025-07-21:
> Details please. Which version of epics-base? Which environment? (looks like mingw on windows?)

OK, epics-base version is 7.0.9，my computer system is windows 10 64bit , and compile tool is mingw64, details like this:

[image]

gtt875, 2025-07-22:
> I am confused since it should always be?

> Based on the output above (which I can not quote since it is an image) I think the issue is that `-DUSE_TYPED_DSET` is not being passed to the compiler. I am confused since it should always be?
> 
> [epics-base/modules/database/test/ioc/db/Makefile](https://github.com/epics-base/epics-base/blob/f29137b98532233c6154bf88defd5f1d203962e9/modules/database/test/ioc/db/Makefile#L16)
> 
> Line 16 in [f29137b](/epics-base/epics-base/commit/f29137b98532233c6154bf88defd5f1d203962e9)
> 
>  USR_CPPFLAGS += -DUSE_TYPED_RSET -DUSE_TYPED_DSET 
> [@gtt875](https://github.com/gtt875) Are you explicitly setting or appending the `USR_CPPFLAGS` make variable yourself?

Could you please provide an operation guide for compiling and installing epics-base 7.0.9 on a Windows 10 64-bit system？
I find it quite challenging to compile on Windows 10 64-bit. Are there any specific requirements for the compiler version and environment configuration?Today, I found the following errors during compilation：

[image]
"true" is used as a variable name, but it is a keyword, resulting in a compilation error.
So I changed "true" to "enable",This problem has been solved, but we encountered the following problem：

[image]
So I found the source code：

[image]

[image]
But I don't known why?please help me, thanks!
I use “MSYS2” as my package management tool, and the compiler toolchain I use is mingw-w64-x86_64-toolchain，  my gcc version and configuration like this:

[image]

The complete compilation log is as follows:

[log.txt](https://github.com/user-attachments/files/21363541/log.txt)

mdavidsaver, 2025-07-22:
> "true" is used as a variable name, but it is a keyword

You must have a newer version of mingw.  This error was fixed after the 7.0.9 release with f827bd7e34450ffbfa6e1d724e7164c24117bc2d.  Please re-test with the latest from the 7.0 branch.

> The complete compilation log is as follows:

Could you expand this log to start from an empty directory, first git-clone, then build.

In particular, are you editing any files after cloning, or setting any compiler or EPICS related environment variables?

> Could you please provide an operation guide for compiling and installing epics-base 7.0.9 on a Windows 10 64-bit system？

I personally can not.  I have used Linux exclusively for many years, and am happier for it!

epics-base uses github-actions continuous integration, with a mingw cross build job, and a native MSVC job.  So we know these targets can build, at least for the tested compiler and OS versions.  See `.github/workflows/ci-scripts-build.yml`.

closed 2025-07-25 by mdavidsaver