epics-base/epics-base #694: Created an IOC example success，But compile error

gtt875, 2025-07-29:
After successfully compiling epicsbase7 using mingw-x64 on Windows 64-bit, I created an IOC example following the instructions. However, when I tried to compile this IOC example using the make command in the same environment, I encountered the following error:
[image]

I use last version code:

[image]

anjohnson, 2025-07-30:
Core Group: This error probably demonstrates that our CI builds don't test compiling the template applications. One of the members will add the changes that we think will fix your compile problem, and we will take this issue as a suggestion to add template builds to our main Base ci instructions.

gtt875, 2025-07-31:
> This seems related to this issue here: [#574](https://github.com/epics-base/epics-base/issues/574)
> 
> If you modify the Makefile in the source directory from
> 
> # Use typed rset structure (see 3.16.1 release notes)
> USR_CPPFLAGS += -DUSE_TYPED_RSET
> to become
> 
> # Use typed rset structure (see 3.16.1 release notes)
> USR_CPPFLAGS += -DUSE_TYPED_RSET
> USR_CPPFLAGS += -DUSE_TYPED_DSET
> (note the DSET vs RSET) then it should compile correctly.

The meanings of these two ways of writing should be equivalent！

anjohnson, 2025-07-31:
That wasn't quite the fix that I thought would be needed, glad you found it — thanks. We'll fix the templates properly before the next release.

@Group: The device support was still using K&R function declarations, not ANSI function prototypes!